Linutronix/elbe #410: Using pbuilder with noauth Tag end with unexpected token `newlinie`

danielfli, 2024-08-07:
Hi, 

the clean up work from this [Commit ](https://github.com/Linutronix/elbe/commit/e3bf8851af8797cbda458f5f93edf92470341d04) doesn't work with the special pbuilderrc file rules.

https://github.com/Linutronix/elbe/blame/c3246bd329003b515343b97b963db36aa33f5b83/elbepack/pbuilder.py#L37C10-L37C10

```python
fp.write("""for i in "${!DEBOOTSTRAPOPTS[@]}"; do if [[ ${DEBOOTSTRAPOPTS[i]}
                 == "--force-check-gpg" ]]; then unset 'DEBOOTSTRAPOPTS[i]'; break; fi done\n""")
```

If I remove the line break like this following, it works fine. 

```python
fp.write("""for i in "${!DEBOOTSTRAPOPTS[@]}"; do if [[ ${DEBOOTSTRAPOPTS[i]} == "--force-check-gpg" ]]; then unset 'DEBOOTSTRAPOPTS[i]'; break; fi done\n""")
```

[image]

t-8ch, 2024-08-07:
Makes sense, thanks for the report.

t-8ch, 2024-08-07:
Could you test this patch?
https://lists.linutronix.de/pipermail/elbe-devel/2024-August/007367.html

danielfli, 2024-08-07:
Looks good to me. The pbuilderrc file is accepted of the pbuilder cmd. 

But now is the pbuilder step hang on the hook file `pbuilder/hooks.d/G10elbe_apt_sources ` for the local repo. 

```bash
#!/bin/sh
cat -> /etc/apt/sources.list <<EOF
deb http://127.0.0.1:7588/repo/var/cache/elbe/01912d13-374b-7268-a8d9-474f88d4e44b/repo bookworm main
deb [trusted=yes arch=amd64] http://1x.x.x.x:8081/mirror/debian bookworm main
deb [trusted=yes] http://1xx.xx.x.x:8081/repo/ bookworm main
EOF
cat -> /etc/apt/trusted.gpg.d/elbe-localrepo.asc <<EOF
-----BEGIN PGP PUBLIC KEY BLOCK-----

mQENBGazeagBCACksOcWf8iPv5i+cbWN4sL8kXrY1x7q4ZC+eMguaKGIPWBtTCEA
```

How can I add the "[trusted=true]" sign to the local Repo `http://127.0.0.1:7588/repo/var/cache/elbe/01912d13-374b-7268-a8d9-474f88d4e44b/repo` ?

t-8ch, 2024-08-07:
Are you running from git?
IIRC the local repo should have a working key setup.

closed 2024-08-08 by t-8ch

t-8ch, 2024-08-08:
c105bf6fc42a1369cb621b3ebeded040578e9237 should be the correct fix.

( d7280698104189293fed09c6a4a9a13cbcc8436f may be also required )

> Is it alright if I do it like this?

I'm not sure why [trusted=yes] does silence this error, to be honest.

danielfli, 2024-08-08:
Okay I did a double check, and it seems to be wrong, my test workspace was wrong. 

That's wrong:
> If I add a "[trusted=true]" sign to the G10elbe_apt_sources file like the follow and execute the pbuilder inside the initvm workspace, it ends with no error.

I get with [trusted=yes] still the 404 error to the local repo.

t-8ch, 2024-08-08:
Does it work with the proposed changes?

danielfli, 2024-08-08:
It works after these changes. Thank you!

> https://github.com/Linutronix/elbe/commit/c105bf6fc42a1369cb621b3ebeded040578e9237 should be the correct fix.

> ( https://github.com/Linutronix/elbe/commit/d7280698104189293fed09c6a4a9a13cbcc8436f may be also required )

t-8ch, 2024-08-08:
Thanks for confirming. I'm working on a proper bugfix release.

danielfli, 2024-08-08:
That sounds fantastic. Thanks!

commit 310db41c (2024-08-08): elbepack: pbuilder: fix shell loop in config file

Commit e3bf8851af87 ("Clean up the code to match python PEP8 standard")
added a linebreak in the middle of a conditional, which is invalid syntax.

Make sure that the condition stays up on one line in the config
file, while still maintaining resonable source line length.

Closes #410