Linutronix/elbe #411: README file instructions are not working as expected.

viveky1794, 2024-08-18:
Hi ELBE team,
I tried to make debian rootfs for my beaglebone black board. But it looks like README file missed sometime. 
I am not getting why am I getting this error ?
I tried all the steps from your Quick start guide.

`
ninja@ninja:~/BBB/debian-rootfs/elbe$ cat /sys/module/kvm_intel/parameters/nested
Y
ninja@ninja:~/BBB/debian-rootfs/elbe$ lsb_release -a
No LSB modules are available.
Distributor ID: Ubuntu
Description:    Ubuntu 24.04 LTS
Release:        24.04
Codename:       noble
ninja@ninja:~/BBB/debian-rootfs/elbe$ cat /etc/os-release
PRETTY_NAME="Ubuntu 24.04 LTS"
NAME="Ubuntu"
VERSION_ID="24.04"
VERSION="24.04 LTS (Noble Numbat)"
VERSION_CODENAME=noble
ID=ubuntu
ID_LIKE=debian
HOME_URL="https://www.ubuntu.com/"
SUPPORT_URL="https://help.ubuntu.com/"
BUG_REPORT_URL="https://bugs.launchpad.net/ubuntu/"
PRIVACY_POLICY_URL="https://www.ubuntu.com/legal/terms-and-policies/privacy-policy"
UBUNTU_CODENAME=noble
LOGO=ubuntu-logo
ninja@ninja:~/BBB/debian-rootfs/elbe$ 
ninja@ninja:~/BBB/debian-rootfs/elbe$ rm -rf initvm
ninja@ninja:~/BBB/debian-rootfs/elbe$ ./elbe initvm create ./examples/armhf-ti-beaglebone-black.xml 
Import ubuntu-keyring-2018-archive.gpg: 
gpg: keybox '/tmp/tmpb3l6zz9s/pubring.gpg' created
gpg: key 871920D1991BC93C: public key "Ubuntu Archive Automatic Signing Key (2018) <ftpmaster@ubuntu.com>" imported
gpg: Total number processed: 1
gpg:               imported: 1
Import ubuntu-keyring-2012-cdimage.gpg: 
gpg: key D94AA3F0EFE21092: public key "Ubuntu CD Image Automatic Signing Key (2012) <cdimage@ubuntu.com>" imported
gpg: Total number processed: 1
gpg:               imported: 1
--2024-08-18 11:13:57--  http://deb.debian.org/debian/dists/bookworm/Release
Resolving deb.debian.org (deb.debian.org)... 151.101.158.132, 2a04:4e42:25::644
Connecting to deb.debian.org (deb.debian.org)|151.101.158.132|:80... connected.
HTTP request sent, awaiting response... 200 OK
Length: 149225 (146K)
Saving to: ‘/tmp/tmpb3l6zz9s/Release’

/tmp/tmpb3l6zz9s/Release           100%[================================================================>] 145.73K  --.-KB/s    in 0.04s   

2024-08-18 11:13:57 (3.24 MB/s) - ‘/tmp/tmpb3l6zz9s/Release’ saved [149225/149225]

Signature with unknown key: A7236886F3CCCAAD148A27F80E98404D386FA1D9
Signature with unknown key: 4CB50190207B4758A3F73A796ED0E7B82643E131
Signature with unknown key: 4D64FEC119C2029067D6E791F8D2585B8783D481
[ERROR]Failure to download kernel/initrd debian Package:
[ERROR]
[ERROR]InvalidSignature Failed to verify Release file
[ERROR]
[ERROR]Check Mirror configuration
'elbe init' Failed
Giving up
ninja@ninja:~/BBB/debian-rootfs/elbe$ git log --oneline -2
a1c6a725 (HEAD -> newBranch, tag: v15.0, origin/releases/v15.0) Revert "release: back to development"
6804e552 release: back to development
ninja@ninja:~/BBB/debian-rootfs/elbe$ 
`

Thanks & Regards
Vivek Yadav

t-8ch, 2024-08-19:
Hi,

please try to install the package `debian-archive-keyring`.

t-8ch, 2024-08-23:
@peccehu 

The error message "Network is unreachable" seems to be fairly clear.

It seems you are using a proxy, is that configured correctly?

peccehu, 2024-08-23:
Yes, I'm behind a proxy, but I thought this was properly configured. In /etc/apt/apt.conf  (for apt), in /etc/profile, in /etc/wgetrc (for wget). Apt works fine when installing and updating.
But it's a fresh install of bookworm; so maybe I'm overlooking something? Another proxy setting somewhere?
Thank you for the feedback.

t-8ch, 2024-08-23:
Which version of elbe are you using exactly?
Try to set the `http_proxy` env variable.
Elbe does not use the system `apt.conf` (and `wgetrc` only coincidentally)

peccehu, 2024-08-23:
I installed elbe today on a fresh bookworm installation; following the instructions on the page https://elbe-rfs.org/download/ for bookworm.
The version which is installed is 15.0
The env variable http_proxy is properly set (that was done in /etc/profile)

peccehu, 2024-08-23:
My issue is solved (different from the initial issue; sorry for hijacking the thread).
I had to set HTTP_PROXY  (capitals, not lower case) as in 
`export HTTP_PROXY=10.10.1.10:3128`

kind regards,

peccehu, 2024-08-23:
Unfortunately still not solved.
Now I'm having troube to contact to Debian archive mirror later on in the initvm creation process.
Should I specify the proxy server on the command line ?

```
devel@bookworm:~/SVN/LINUXHC3/trunk/platform/bookworm$ elbe initvm create
Import debian-archive-bullseye-automatic.asc:
gpg: keybox '/tmp/tmpl4_uyy6p/pubring.gpg' created
gpg: key 73A4F27B8DD47936: 5 signatures not checked due to missing keys
gpg: key 73A4F27B8DD47936: public key "Debian Archive Automatic Signing Key (11/bullseye) <ftpmaster@debian.org>" imported
gpg: Total number processed: 1
gpg:               imported: 1
Import debian-archive-buster-automatic.asc:
gpg: key DC30D7C23CBBABEE: 4 signatures not checked due to missing keys
gpg: key DC30D7C23CBBABEE: public key "Debian Archive Automatic Signing Key (10/buster) <ftpmaster@debian.org>" imported
gpg: Total number processed: 1
gpg:               imported: 1
Import debian-archive-bookworm-security-automatic.asc:
gpg: key 254CF3B5AEC0A8F0: 4 signatures not checked due to missing keys
gpg: key 254CF3B5AEC0A8F0: public key "Debian Security Archive Automatic Signing Key (12/bookworm) <ftpmaster@debian.org>" imported
gpg: Total number processed: 1
gpg:               imported: 1
Import debian-archive-bookworm-stable.asc:
gpg: key F8D2585B8783D481: public key "Debian Stable Release Key (12/bookworm) <debian-release@lists.debian.org>" imported
gpg: Total number processed: 1
gpg:               imported: 1
Import debian-archive-bookworm-automatic.asc:
gpg: key B7C5D7D6350947F8: 3 signatures not checked due to missing keys
gpg: key B7C5D7D6350947F8: public key "Debian Archive Automatic Signing Key (12/bookworm) <ftpmaster@debian.org>" imported
gpg: Total number processed: 1
gpg:               imported: 1
Import debian-archive-buster-stable.asc:
gpg: key DCC9EFBF77E11517: public key "Debian Stable Release Key (10/buster) <debian-release@lists.debian.org>" imported
gpg: Total number processed: 1
gpg:               imported: 1
Import debian-archive-buster-security-automatic.asc:
gpg: key 4DFAB270CAA96DFA: 4 signatures not checked due to missing keys
gpg: key 4DFAB270CAA96DFA: public key "Debian Security Archive Automatic Signing Key (10/buster) <ftpmaster@debian.org>" imported
gpg: Total number processed: 1
gpg:               imported: 1
Import debian-archive-bullseye-security-automatic.asc:
gpg: key A48449044AAD5C5D: 2 signatures not checked due to missing keys
gpg: key A48449044AAD5C5D: public key "Debian Security Archive Automatic Signing Key (11/bullseye) <ftpmaster@debian.org>" imported
gpg: Total number processed: 1
gpg:               imported: 1
Import debian-archive-bullseye-stable.asc:
gpg: key 605C66F00D6C9793: 3 signatures not checked due to missing keys
gpg: key 605C66F00D6C9793: public key "Debian Stable Release Key (11/bullseye) <debian-release@lists.debian.org>" imported
gpg: Total number processed: 1
gpg:               imported: 1
--2024-08-23 07:07:46--  http://deb.debian.org/debian/dists/bookworm/Release
Connecting to 192.168.40.17:3128... connected.
Proxy request sent, awaiting response... 200 OK
Length: 149225 (146K)
Saving to: ‘/tmp/tmpl4_uyy6p/Release’

/tmp/tmpl4_uyy6p/Release                                    100%[===========================================================================================================================================>] 145.73K  --.-KB/s    in 0.06s

2024-08-23 07:07:47 (2.30 MB/s) - ‘/tmp/tmpl4_uyy6p/Release’ saved [149225/149225]

Debian Archive Automatic Signing Key <ftpmaster@debian.org> (A7236886F3CCCAAD148A27F80E98404D386FA1D9):
VALID (Untrusted).
Debian Archive Automatic Signing Key <ftpmaster@debian.org> (4CB50190207B4758A3F73A796ED0E7B82643E131):
VALID (Untrusted).
Debian Stable Release Key <debian-release@lists.debian.org> (4D64FEC119C2029067D6E791F8D2585B8783D481):
VALID (Untrusted).
--2024-08-23 07:07:47--  http://deb.debian.org/debian/dists/bookworm/main/installer-amd64/current/images/SHA256SUMS
Connecting to 192.168.40.17:3128... connected.
Proxy request sent, awaiting response... 200 OK
Length: 78509 (77K)
[... 121 lines cut ...]
Detecting network hardware  ... 2%... 95%... 100%
Detecting link on ens4; please wait...  ... 0%
Waiting for link-local address...  ... 16%... 25%... 33%... 41%... 50%... 100%
Attempting IPv6 autoconfiguration...  ... 8%... 100%
Attempting IPv6 autoconfiguration...  ... 5%... 10%... 20%... 30%... 40%... 50%... 60%... 70%... 80%... 90%... 100%
Configuring the network with DHCP  ... 4%... 100%
Checking the Debian archive mirror  ... 25%... 50%... 75%... 100%... 100%
Choose a mirror of the Debian archive
-------------------------------------

!! ERROR: Bad archive mirror

An error has been detected while trying to use the specified Debian archive
mirror.

Possible reasons for the error are: incorrect mirror specified; mirror is not
available (possibly due to an unreliable network connection); mirror is broken
(for example because an invalid Release file was found); mirror does not
support the correct Debian version.

Additional details may be available in /var/log/syslog or on virtual console 4.

Please check the specified mirror or try a different one.
[Press enter to continue]
```

t-8ch, 2024-08-23:
Can you look at `initvm/installer.log`?

peccehu, 2024-08-23:
root@bookworm:/home/devel/SVN/LINUXHC3/trunk/platform/bookworm/initvm#
root@bookworm:/home/devel/SVN/LINUXHC3/trunk/platform/bookworm/initvm# cat installer.log  |tail
Aug 23 12:34:05 main-menu[2685]: INFO: Falling back to the package description for brltty-udeb
Aug 23 12:34:05 main-menu[2685]: INFO: Menu item 'choose-mirror' selected
Aug 23 12:34:05 anna-install: Queueing udeb apt-mirror-setup for later installation
Aug 23 12:34:05 choose-mirror[3619]: DEBUG: command: wget --no-verbose http://deb.debian.org/debian/dists/oldstable/Release -O - | grep -E '^(Suite|Codename|Architectures):'
Aug 23 12:35:22 choose-mirror[3619]: DEBUG: command: wget --no-verbose http://deb.debian.org/debian/dists/stable/Release -O - | grep -E '^(Suite|Codename|Architectures):'
Aug 23 12:36:39 choose-mirror[3619]: DEBUG: command: wget --no-verbose http://deb.debian.org/debian/dists/testing/Release -O - | grep -E '^(Suite|Codename|Architectures):'
Aug 23 12:37:56 choose-mirror[3619]: DEBUG: command: wget --no-verbose http://deb.debian.org/debian/dists/unstable/Release -O - | grep -E '^(Suite|Codename|Architectures):'
Aug 23 12:39:12 choose-mirror[3619]: INFO: mirror does not have any suite symlinks
Aug 23 12:39:12 choose-mirror[3619]: DEBUG: command: wget --no-verbose http://deb.debian.org/debian/dists/bookworm/Release -O - | grep -E '^(Suite|Codename|Architectures):'
Aug 23 12:40:29 choose-mirror[3619]: WARNING **: mirror does not support the specified release (bookworm)
root@bookworm:/home/devel/SVN/LINUXHC3/trunk/platform/bookworm/initvm#

t-8ch, 2024-08-23:
Please try to use the `primary_proxy` xml element.
(This should be more automatic...)

peccehu, 2024-08-23:
I added the following section in my target XML file 

```
    <target>
		<mirror>
		  <primary_proxy>http://192.168.40.17:3128</primary_proxy>
		</mirror>
        <hostname>hc3</hostname>

```
and then ran again the initvm creation with the target XLM as argument 
But the result is the same as previous post.

`devel@bookworm:~/SVN/LINUXHC3/trunk/platform/bookworm$ elbe initvm create hc3prod.xml`

t-8ch, 2024-08-23:
Sorry, this should be part of the `<initvm>` tag, which by default is taken from `elbepack/init/default-init.xml`, but you can specify your own in your elbe xml.