Linutronix/elbe #412: Issues with some additional repositories and keys

martin-29, 2024-08-27:
With version 15.0 we ran into a problem we didn't have with 14.9.3.

We add some additional repositories, like from Docker and Mender, to our url-list.
To properly work with these repositories we have added the raw keys like shown in the snippet below.

```xml
<url>
	<binary>https://download.docker.com/linux/debian bookworm stable</binary>
	<raw-key>
		-----BEGIN PGP PUBLIC KEY BLOCK-----

		mQINBFit2ioBEADhWpZ8/wvZ6hUTiXOwQHXMAlaFHcPH9hAtr4F1y2+OYdbtMuth
		lqqwp028AqyY+PRfVMtSYMbjuQuu5byyKR01BbqYhuS3jtqQmljZ/bJvXqnmiVXh
		38UuLa+z077PxyxQhu5BbqntTPQMfiyqEiU+BKbq2WmANUKQf+1AmZY/IruOXbnq
		L4C1+gJ8vfmXQt99npCaxEjaNRVYfOS8QcixNzHUYnb6emjlANyEVlZzeqo7XKl7
		UrwV5inawTSzWNvtjEjj4nJL8NsLwscpLPQUhTQ+7BbQXAwAmeHCUTQIvvWXqw0N
		cmhh4HgeQscQHYgOJjjDVfoY5MucvglbIgCqfzAHW9jxmRL4qbMZj+b1XoePEtht
		ku4bIQN1X5P07fNWzlgaRL5Z4POXDDZTlIQ/El58j9kp4bnWRCJW0lya+f8ocodo
		vZZ+Doi+fy4D5ZGrL4XEcIQP/Lv5uFyf+kQtl/94VFYVJOleAv8W92KdgDkhTcTD
        	G7c0tIkVEKNUq48b3aQ64NOZQW7fVjfoKwEZdOqPE72Pa45jrZzvUFxSpdiNk2tZ
		XYukHjlxxEgBdC/J3cMMNRE1F4NCA3ApfV1Y7/hTeOnmDuDYwr9/obA8t016Yljj
	        q5rdkywPf4JF8mXUW5eCN1vAFHxeg9ZWemhBtQmGxXnw9M+z6hWwc6ahmwARAQAB
		tCtEb2NrZXIgUmVsZWFzZSAoQ0UgZGViKSA8ZG9ja2VyQGRvY2tlci5jb20+iQI3
		BBMBCgAhBQJYrefAAhsvBQsJCAcDBRUKCQgLBRYCAwEAAh4BAheAAAoJEI2BgDwO
		v82IsskP/iQZo68flDQmNvn8X5XTd6RRaUH33kXYXquT6NkHJciS7E2gTJmqvMqd
		tI4mNYHCSEYxI5qrcYV5YqX9P6+Ko+vozo4nseUQLPH/ATQ4qL0Zok+1jkag3Lgk
		jonyUf9bwtWxFp05HC3GMHPhhcUSexCxQLQvnFWXD2sWLKivHp2fT8QbRGeZ+d3m
		6fqcd5Fu7pxsqm0EUDK5NL+nPIgYhN+auTrhgzhK1CShfGccM/wfRlei9Utz6p9P
		XRKIlWnXtT4qNGZNTN0tR+NLG/6Bqd8OYBaFAUcue/w1VW6JQ2VGYZHnZu9S8LMc
		FYBa5Ig9PxwGQOgq6RDKDbV+PqTQT5EFMeR1mrjckk4DQJjbxeMZbiNMG5kGECA8
		g383P3elhn03WGbEEa4MNc3Z4+7c236QI3xWJfNPdUbXRaAwhy/6rTSFbzwKB0Jm
		ebwzQfwjQY6f55MiI/RqDCyuPj3r3jyVRkK86pQKBAJwFHyqj9KaKXMZjfVnowLh
		9svIGfNbGHpucATqREvUHuQbNnqkCx8VVhtYkhDb9fEP2xBu5VvHbR+3nfVhMut5
		G34Ct5RS7Jt6LIfFdtcn8CaSas/l1HbiGeRgc70X/9aYx/V/CEJv0lIe8gP6uDoW
		FPIZ7d6vH+Vro6xuWEGiuMaiznap2KhZmpkgfupyFmplh0s6knymuQINBFit2ioB
		EADneL9S9m4vhU3blaRjVUUyJ7b/qTjcSylvCH5XUE6R2k+ckEZjfAMZPLpO+/tF
		M2JIJMD4SifKuS3xck9KtZGCufGmcwiLQRzeHF7vJUKrLD5RTkNi23ydvWZgPjtx
		Q+DTT1Zcn7BrQFY6FgnRoUVIxwtdw1bMY/89rsFgS5wwuMESd3Q2RYgb7EOFOpnu
		w6da7WakWf4IhnF5nsNYGDVaIHzpiqCl+uTbf1epCjrOlIzkZ3Z3Yk5CM/TiFzPk
		z2lLz89cpD8U+NtCsfagWWfjd2U3jDapgH+7nQnCEWpROtzaKHG6lA3pXdix5zG8
		eRc6/0IbUSWvfjKxLLPfNeCS2pCL3IeEI5nothEEYdQH6szpLog79xB9dVnJyKJb
		VfxXnseoYqVrRz2VVbUI5Blwm6B40E3eGVfUQWiux54DspyVMMk41Mx7QJ3iynIa
		1N4ZAqVMAEruyXTRTxc9XW0tYhDMA/1GYvz0EmFpm8LzTHA6sFVtPm/ZlNCX6P1X
		zJwrv7DSQKD6GGlBQUX+OeEJ8tTkkf8QTJSPUdh8P8YxDFS5EOGAvhhpMBYD42kQ
		pqXjEC+XcycTvGI7impgv9PDY1RCC1zkBjKPa120rNhv/hkVk/YhuGoajoHyy4h7
		ZQopdcMtpN2dgmhEegny9JCSwxfQmQ0zK0g7m6SHiKMwjwARAQABiQQ+BBgBCAAJ
		BQJYrdoqAhsCAikJEI2BgDwOv82IwV0gBBkBCAAGBQJYrdoqAAoJEH6gqcPyc/zY
		1WAP/2wJ+R0gE6qsce3rjaIz58PJmc8goKrir5hnElWhPgbq7cYIsW5qiFyLhkdp
		YcMmhD9mRiPpQn6Ya2w3e3B8zfIVKipbMBnke/ytZ9M7qHmDCcjoiSmwEXN3wKYI
		mD9VHONsl/CG1rU9Isw1jtB5g1YxuBA7M/m36XN6x2u+NtNMDB9P56yc4gfsZVES
		KA9v+yY2/l45L8d/WUkUi0YXomn6hyBGI7JrBLq0CX37GEYP6O9rrKipfz73XfO7
		JIGzOKZlljb/D9RX/g7nRbCn+3EtH7xnk+TK/50euEKw8SMUg147sJTcpQmv6UzZ
		cM4JgL0HbHVCojV4C/plELwMddALOFeYQzTif6sMRPf+3DSj8frbInjChC3yOLy0
		6br92KFom17EIj2CAcoeq7UPhi2oouYBwPxh5ytdehJkoo+sN7RIWua6P2WSmon5
		U888cSylXC0+ADFdgLX9K2zrDVYUG1vo8CX0vzxFBaHwN6Px26fhIT1/hYUHQR1z
		VfNDcyQmXqkOnZvvoMfz/Q0s9BhFJ/zU6AgQbIZE/hm1spsfgvtsD1frZfygXJ9f
		irP+MSAI80xHSf91qSRZOj4Pl3ZJNbq4yYxv0b1pkMqeGdjdCYhLU+LZ4wbQmpCk
		SVe2prlLureigXtmZfkqevRz7FrIZiu9ky8wnCAPwC7/zmS18rgP/17bOtL4/iIz
		QhxAAoAMWVrGyJivSkjhSGx1uCojsWfsTAm11P7jsruIL61ZzMUVE2aM3Pmj5G+W
		9AcZ58Em+1WsVnAXdUR//bMmhyr8wL/G1YO1V3JEJTRdxsSxdYa4deGBBY/Adpsw
		24jxhOJR+lsJpqIUeb999+R8euDhRHG9eFO7DRu6weatUJ6suupoDTRWtr/4yGqe
		dKxV3qQhNLSnaAzqW/1nA3iUB4k7kCaKZxhdhDbClf9P37qaRW467BLCVO/coL3y
		Vm50dwdrNtKpMBh3ZpbB1uJvgi9mXtyBOMJ3v8RZeDzFiG8HdCtg9RvIt/AIFoHR
		H3S+U79NT6i0KPzLImDfs8T7RlpyuMc4Ufs8ggyg9v3Ae6cN3eQyxcK3w0cbBwsh
	        /nQNfsA6uu+9H7NhbehBMhYnpNZyrHzCmzyXkauwRAqoCbGCNykTRwsur9gS41TQ
		M8ssD1jFheOJf3hODnkKU+HKjvMROl1DK7zdmLdNzA1cvtZH/nCC9KPj1z8QC47S
		xx+dTZSx4ONAhwbS/LN3PoKtn8LPjY9NP9uDWI+TWYquS2U+KHDrBDlsgozDbs/O
		jCxcpDzNmXpWQHEtHU7649OXHP7UeNST1mCUCH5qdank0V1iejF6/CfTFU4MfcrG
		YT90qFF93M3v01BbxP+EIY2/9tiIPbrd
		=0YYh
		-----END PGP PUBLIC KEY BLOCK-----
	</raw-key>
</url>
```

With version 14.9.3 this worked just fine, but now we get following error:

```log
[ERROR]Failed to build the Apt Cache.
Probable cause might be:
  - Problems with internet connection
  - Broken mirrors
Traceback (most recent call last):
  File "/usr/lib/python3/dist-packages/elbepack/elbeproject.py", line 1011, in install_packages
    self.get_rpcaptcache(env=target).update()
  File "<string>", line 2, in update
  File "/usr/lib/python3.11/multiprocessing/managers.py", line 837, in _callmethod
    raise convert_to_error(kind, result)
apt.cache.FetchFailedException: W:GPG error: http://downloads.mender.io/repos/debian debian/bookworm/stable InRelease: The following signatures couldn't be verified because the public key is not available: NO_PUBKEY 24072B80A1B29B00, E:The repository 'http://downloads.mender.io/repos/debian debian/bookworm/stable InRelease' is not signed., W:Updating from such a repository can't be done securely, and is therefore disabled by default., W:See apt-secure(8) manpage for repository creation and user configuration details., W:GPG error: https://download.docker.com/linux/debian bookworm InRelease: The following signatures couldn't be verified because the public key is not available: NO_PUBKEY 7EA0A9C3F273FCD8, E:The repository 'http://download.docker.com/linux/debian bookworm InRelease' is not signed.

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/usr/lib/python3/dist-packages/elbepack/asyncworker.py", line 171, in execute
    self.project.build(skip_pkglist=False,
  File "/usr/lib/python3/dist-packages/elbepack/elbeproject.py", line 592, in build
    self.install_packages(self.buildenv)
  File "/usr/lib/python3/dist-packages/elbepack/elbeproject.py", line 1013, in install_packages
    raise AptCacheUpdateError(e)
elbepack.elbeproject.AptCacheUpdateError: Error Updating rpcaptcache: W:GPG error: http://downloads.mender.io/repos/debian debian/bookworm/stable InRelease: The following signatures couldn't be verified because the public key is not available: NO_PUBKEY 24072B80A1B29B00, E:The repository 'http://downloads.mender.io/repos/debian debian/bookworm/stable InRelease' is not signed., W:Updating from such a repository can't be done securely, and is therefore disabled by default., W:See apt-secure(8) manpage for repository creation and user configuration details., W:GPG error: https://download.docker.com/linux/debian bookworm InRelease: The following signatures couldn't be verified because the public key is not available: NO_PUBKEY 7EA0A9C3F273FCD8, E:The repository 'http://download.docker.com/linux/debian bookworm InRelease' is not signed.
Project build was not successful, current status: build_failed
elbe control wait_busy Failed
```

We have also our own APT repository running for custom packages and here we don't have any issues.

Our current workaround is to add the trusted option as shown below, which we want to avoid.

```xml
<url>
	<binary>https://download.docker.com/linux/debian bookworm stable</binary>
	<options>
		<option>trusted=yes</option>
	</options>
	<raw-key>
                 ...
	</raw-key>
</url>
```

We have already checked the contents of `/etc/apt/trusted.gpg.d` to see if the key is added, and it is.
So on the running system an `apt update` works just fine, even if the the trusted option is removed from the `source.list` again.

```sh
root@test:/etc/apt/trusted.gpg.d# ls -l
total 92
total 96
-rw-r--r-- 1 root root 11861 30. Jul 2023  debian-archive-bookworm-automatic.asc
-rw-r--r-- 1 root root 11873 30. Jul 2023  debian-archive-bookworm-security-automatic.asc
-rw-r--r-- 1 root root   461 30. Jul 2023  debian-archive-bookworm-stable.asc
-rw-r--r-- 1 root root 11861 30. Jul 2023  debian-archive-bullseye-automatic.asc
-rw-r--r-- 1 root root 11873 30. Jul 2023  debian-archive-bullseye-security-automatic.asc
-rw-r--r-- 1 root root  3403 30. Jul 2023  debian-archive-bullseye-stable.asc
-rw-r--r-- 1 root root 11093 30. Jul 2023  debian-archive-buster-automatic.asc
-rw-r--r-- 1 root root 11105 30. Jul 2023  debian-archive-buster-security-automatic.asc
-rw-r--r-- 1 root root  1704 30. Jul 2023  debian-archive-buster-stable.asc
-rw-r--r-- 1 root root  2293 26. Aug 14:08 elbe-xml-raw-key3.gpg
-rw-r--r-- 1 root root  2760 26. Aug 14:08 elbe-xml-raw-key4.gpg
-rw-r--r-- 1 root root  1754 26. Aug 14:08 elbe-xml-raw-key5.gpg
```

Any idea why this happens?

t-8ch, 2024-08-27:
@martin-29 Thanks for the report. I don't have any immediate idea, and can't reproduce it.
Do you have an example elbe XML and invocation?

martin-29, 2024-08-27:
Thanks for the quick response.

Sure, I added an example XML with which the issue occur.
The workaround is commented out in the example.

We trigger the build with the following command:

`elbe initvm submit --output /output/test/ ./Export.xml`

```xml
<ns0:RootFileSystem xmlns:ns0="https://www.linutronix.de/projects/Elbe" xmlns:xi="http://www.w3.org/2003/XInclude" xmlns:xsi="http://www.w3.org/2001/XMLSchema-instance" created="2009-05-20T08:50:56" revision="6" xsi:schemaLocation="https://www.linutronix.de/projects/Elbe dbsfed.xsd">
	<project>
		<name>Raspberry Pi 64bit</name>
		<version>1.0.0_dev</version>
		<description>
			Testbuild for 15.0
		</description>
		<buildtype>aarch64</buildtype>
		<mirror>
			<primary_host>ftp.debian.org</primary_host>
			<primary_path>/debian</primary_path>
			<primary_proto>http</primary_proto>
			<url-list>
				<url>
					<binary>http://ftp.debian.org/debian bookworm contrib non-free non-free-firmware</binary>
					<source>http://ftp.debian.org/debian bookworm contrib non-free non-free-firmware</source>
				</url>
				<url>
					<binary>http://ftp.debian.org/debian bookworm-backports contrib main non-free non-free-firmware</binary>
					<source>http://ftp.debian.org/debian bookworm-backports contrib main non-free non-free-firmware</source>
				</url>
				<url>
					<binary>http://security.debian.org/debian-security bookworm-security contrib main non-free non-free-firmware</binary>
					<source>http://security.debian.org/debian-security bookworm-security contrib main non-free non-free-firmware</source>
				</url>
				<url>
					<binary>https://download.docker.com/linux/debian bookworm stable</binary>
					<!-- Workaround -->
					<!-- <options>
						<option>trusted=yes</option>
					</options> -->
					<raw-key>
						-----BEGIN PGP PUBLIC KEY BLOCK-----

						mQINBFit2ioBEADhWpZ8/wvZ6hUTiXOwQHXMAlaFHcPH9hAtr4F1y2+OYdbtMuth
						lqqwp028AqyY+PRfVMtSYMbjuQuu5byyKR01BbqYhuS3jtqQmljZ/bJvXqnmiVXh
						38UuLa+z077PxyxQhu5BbqntTPQMfiyqEiU+BKbq2WmANUKQf+1AmZY/IruOXbnq
						L4C1+gJ8vfmXQt99npCaxEjaNRVYfOS8QcixNzHUYnb6emjlANyEVlZzeqo7XKl7
						UrwV5inawTSzWNvtjEjj4nJL8NsLwscpLPQUhTQ+7BbQXAwAmeHCUTQIvvWXqw0N
						cmhh4HgeQscQHYgOJjjDVfoY5MucvglbIgCqfzAHW9jxmRL4qbMZj+b1XoePEtht
						ku4bIQN1X5P07fNWzlgaRL5Z4POXDDZTlIQ/El58j9kp4bnWRCJW0lya+f8ocodo
						vZZ+Doi+fy4D5ZGrL4XEcIQP/Lv5uFyf+kQtl/94VFYVJOleAv8W92KdgDkhTcTD
						G7c0tIkVEKNUq48b3aQ64NOZQW7fVjfoKwEZdOqPE72Pa45jrZzvUFxSpdiNk2tZ
						XYukHjlxxEgBdC/J3cMMNRE1F4NCA3ApfV1Y7/hTeOnmDuDYwr9/obA8t016Yljj
						q5rdkywPf4JF8mXUW5eCN1vAFHxeg9ZWemhBtQmGxXnw9M+z6hWwc6ahmwARAQAB
						tCtEb2NrZXIgUmVsZWFzZSAoQ0UgZGViKSA8ZG9ja2VyQGRvY2tlci5jb20+iQI3
						BBMBCgAhBQJYrefAAhsvBQsJCAcDBRUKCQgLBRYCAwEAAh4BAheAAAoJEI2BgDwO
						v82IsskP/iQZo68flDQmNvn8X5XTd6RRaUH33kXYXquT6NkHJciS7E2gTJmqvMqd
						tI4mNYHCSEYxI5qrcYV5YqX9P6+Ko+vozo4nseUQLPH/ATQ4qL0Zok+1jkag3Lgk
						jonyUf9bwtWxFp05HC3GMHPhhcUSexCxQLQvnFWXD2sWLKivHp2fT8QbRGeZ+d3m
						6fqcd5Fu7pxsqm0EUDK5NL+nPIgYhN+auTrhgzhK1CShfGccM/wfRlei9Utz6p9P
						XRKIlWnXtT4qNGZNTN0tR+NLG/6Bqd8OYBaFAUcue/w1VW6JQ2VGYZHnZu9S8LMc
						FYBa5Ig9PxwGQOgq6RDKDbV+PqTQT5EFMeR1mrjckk4DQJjbxeMZbiNMG5kGECA8
						g383P3elhn03WGbEEa4MNc3Z4+7c236QI3xWJfNPdUbXRaAwhy/6rTSFbzwKB0Jm
						ebwzQfwjQY6f55MiI/RqDCyuPj3r3jyVRkK86pQKBAJwFHyqj9KaKXMZjfVnowLh
						9svIGfNbGHpucATqREvUHuQbNnqkCx8VVhtYkhDb9fEP2xBu5VvHbR+3nfVhMut5
						G34Ct5RS7Jt6LIfFdtcn8CaSas/l1HbiGeRgc70X/9aYx/V/CEJv0lIe8gP6uDoW
						FPIZ7d6vH+Vro6xuWEGiuMaiznap2KhZmpkgfupyFmplh0s6knymuQINBFit2ioB
						EADneL9S9m4vhU3blaRjVUUyJ7b/qTjcSylvCH5XUE6R2k+ckEZjfAMZPLpO+/tF
						M2JIJMD4SifKuS3xck9KtZGCufGmcwiLQRzeHF7vJUKrLD5RTkNi23ydvWZgPjtx
[... 264 lines cut ...]
			<!-- External Wifi driver -->
			<pkg>firmware-realtek</pkg>
			<pkg>firmware-atheros</pkg>

			<pkg>network-manager</pkg>
			<pkg>libgpiod2</pkg>

			<!-- Mender packages without sources -->
			<pkg on_src_cd="False">mender-snapshot</pkg>
			<pkg on_src_cd="False">mender-flash</pkg>
			<pkg on_src_cd="False">mender-setup</pkg>
			<pkg on_src_cd="False">mender-auth</pkg>
			<pkg on_src_cd="False">mender-update</pkg>
			<pkg on_src_cd="False">mender-client4</pkg>

			<!-- Docker packages without sources -->
			<pkg on_src_cd="False">containerd.io</pkg>
			<pkg on_src_cd="False">docker-ce</pkg>
			<pkg on_src_cd="False">docker-ce-cli</pkg>
			<pkg on_src_cd="False">docker-compose-plugin</pkg>
		</pkg-list>
	</target>

</ns0:RootFileSystem>
```

t-8ch, 2024-08-27:
The new version of `unarmor_openpgp_keyring()` (which only does an base64 decode) does not work with the RPi key.

martin-29, 2024-08-27:
Ok, but do you know why it doesn't work anymore for Docker and Mender?
I mean I also did a test with a local instance of an APT server (created with reprepro) where I also added the public key on the same way as for Docker and Mender. This doesn't cause any problems and no workaround is needed.

I've just removed this part and the relevant packages from the example, because you it's not easy to reproduce.

Could you reproduce the problem for Docker and Mender?

t-8ch, 2024-08-27:
> Could you reproduce the problem for Docker and Mender?

Not so far.

What is the issue with Docker? I can't see an error message for that.

t-8ch, 2024-08-28:
I can't reproduce the mender/docker issues.
Weirdly the error messages mention http URLs, while the elbe XML file uses https.
Are you using some sort of proxy?

t-8ch, 2024-08-28:
The fix for the RPi repo is here: https://lists.linutronix.de/pipermail/elbe-devel/2024-August/007525.html
You can apply it as follows:

* `ssh -p5022 -o UserKnownHostsFile=/dev/null root@localhost`
* Password `root`
* Apply the patch to `/usr/lib/python3/dist-packages/elbepack/egpg.py`
* `systemctl restart python3-elbe-daemon`

t-8ch, 2024-08-28:
I was able to reproduce the issue with a vanilla bookworm VM.

t-8ch, 2024-08-28:
In my testing, the broken docker/mender repositories where a follow-up error triggered by the bug affecting the RPi repo.
So the patch from https://github.com/Linutronix/elbe/issues/412#issuecomment-2314368918 should fix it.

t-8ch, 2024-08-28:
As a workaround you should be able to use this as key for the RPi repo:
(It's the same as before just with the non-base64 lines removed.)

```
						-----BEGIN PGP PUBLIC KEY BLOCK-----
						mQENBE/d7o8BCACrwqQacGJfn3tnMzGui6mv2lLxYbsOuy/+U4rqMmGEuo3h9m92
						30E2EtypsoWczkBretzLUCFv+VUOxaA6sV9+puTqYGhhQZFuKUWcG7orf7QbZRuu
						TxsEUepW5lg7MExmAu1JJzqM0kMQX8fVyWVDkjchZ/is4q3BPOUCJbUJOsE+kK/6
						8kW6nWdhwSAjfDh06bA5wvoXNjYoDdnSZyVdcYCPEJXEg5jfF/+nmiFKMZBraHwn
						eQsepr7rBXxNcEvDlSOPal11fg90KXpy7Umre1UcAZYJdQeWcHu7X5uoJx/MG5J8
						ic6CwYmDaShIFa92f8qmFcna05+lppk76fsnABEBAAG0IFJhc3BiZXJyeSBQaSBB
						cmNoaXZlIFNpZ25pbmcgS2V5iQE4BBMBAgAiBQJP3e6PAhsDBgsJCAcDAgYVCAIJ
						CgsEFgIDAQIeAQIXgAAKCRCCsSmSf6MwPk6vB/9pePB3IukU9WC9Bammh3mpQTvL
						OifbkzHkmAYxzjfK6D2I8pT0xMxy949+ThzJ7uL60p6T/32ED9DR3LHIMXZvKtuc
						mQnSiNDX03E2p7lIP/htoxW2hDP2n8cdlNdt0M9IjaWBppsbO7IrDppG2B1aRLni
						uD7v8bHRL2mKTtIDLX42Enl8aLAkJYgNWpZyPkDyOqamjijarIWjGEPCkaURF7g4
						d44HvYhpbLMOrz1m6N5Bzoa5+nq3lmifeiWKxioFXU+Hy5bhtAM6ljVb59hbD2ra
						X4+3LXC9oox2flmQnyqwoyfZqVgSQa0B41qEQo8t1bz6Q1Ti7fbMLThmbRHiuQEN
						BE/d7o8BCADNlVtBZU63fm79SjHh5AEKFs0C3kwa0mOhp9oas/haDggmhiXdzeD3
						49JWz9ZTx+vlTq0s+I+nIR1a+q+GL+hxYt4HhxoA6vlDMegVfvZKzqTX9Nr2VqQa
						S4Kz3W5ULv81tw3WowK6i0L7pqDmvDqgm73mMbbxfHD0SyTt8+fk7qX6Ag2pZ4a9
						ZdJGxvASkh0McGpbYJhk1WYD+eh4fqH3IaeJi6xtNoRdc5YXuzILnp+KaJyPE5CR
						qUY5JibOD3qR7zDjP0ueP93jLqmoKltCdN5+yYEExtSwz5lXniiYOJp8LWFCgv5h
						m8aYXkcJS1xVV9Ltno23YvX5edw9QY4hABEBAAGJAR8EGAECAAkFAk/d7o8CGwwA
						CgkQgrEpkn+jMD5Figf/dIC1qtDMTbu5IsI5uZPX63xydaExQNYf98cq5H2fWF6O
						yVR7ERzA2w33hI0yZQrqO6pU9SRnHRxCFvGv6y+mXXXMRcmjZG7GiD6tQWeN/3wb
						EbAn5cg6CJ/Lk/BI4iRRfBX07LbYULCohlGkwBOkRo10T+Ld4vCCnBftCh5x2OtZ
						TOWRULxP36y2PLGVNF+q9pho98qx+RIxvpofQM/842ZycjPJvzgVQsW4LT91KYAE
						4TVf6JjwUM6HZDoiNcX6d7zOhNfQihXTsniZZ6rky287htsWVDNkqOi5T3oTxWUo
						m++/7s3K3L0zWopdhMVcgg6Nt9gcjzqN1c0gy55L/g==
						-----END PGP PUBLIC KEY BLOCK-----
```

martin-29, 2024-08-28:
Now I was also able to make a test with the patch of `elbepack/egpg.py` applied.
With that everything seems to work fine.

But at least I had also problems without the RPI repo inside the url-list. Then the Docker and Mender key still create problems. 
So I think we need to apply the patch to be able to build properly, which shouldn't be a big problem.

Would be good if the next release contains that patch. I think it's in general a good improvement to use the gpg library for dearmoring.

Thanks a lot for your investigation and support!

t-8ch, 2024-08-28:
> Would be good if the next release contains that patch.

Absolutely, that is the plan.

> Thanks a lot for your investigation and support!

You're welcome!

t-8ch, 2024-08-30:
This should be fixed in 15.1 which was just released.

closed 2024-08-30 by t-8ch