Linutronix/elbe #414: Can't use '--qemu' option

FedericoVaga, 2024-09-11:
I'm using ELBE v15.1, and I wanted to experiment with the `--qemu` option. However, I can't do that because of an unrecognized argument.

```
  $ podman run -ti --rm localhost/elbe-builder-image elbe --version
elbe v15.1
  $ podman run -ti --rm localhost/elbe-builder-image elbe --version
elbe v15.1
  $ podman run -ti --rm localhost/elbe-builder-image elbe initvm --help
usage: elbe initvm [-h]

options:
  -h, --help  show this help message and exit
  $ podman run -ti --rm localhost/elbe-builder-image elbe initvm --qemu create
usage: elbe initvm [-h] [-v VARIANTS] [-p PROXY] [--sshport SSHPORT] [--soapport SOAPPORT] [-z GZIP]
                   {start,ensure,stop,destroy,attach,create,submit,sync} ...
elbe initvm: error: unrecognized arguments: --qemu
```

t-8ch, 2024-09-11:
The `--help` message looks broken, I'll look at it.

Try `elbe initvm create --qemu` instead.

FedericoVaga, 2024-09-11:
aaaaah yes, better! Thanks for the tips!

Consider also fixing [the initvm documentation](https://elbe-rfs.org/docs/sphinx/v15.1/elbe-initvm.html) because it says to place `[options]` after `initvm`.

closed 2024-12-19 by t-8ch