Linutronix/elbe #423: policy-rc.d inside build images is blocking service start via invoke-rc.d

martin-29, 2024-12-12:
Using an image created by ELBE v15.2, we discovered the behavior that calling `invoke-rc.d` inside a package postinstall to start a systemd service is blocked by `policy-rc.d`.

This file didn't exist in images created by ELBE 14.9.3 and probably also 15.0.

I understand that this file makes sense during the build process, but afterwards on the resulting image this is in my opinion most of the time not desired.

The problem now is that overwriting or deleting it within the finetuning doesn't work.
Or to be more precise the file just reappears inside the created image.

Inside the log we see that it's being deleted.

```sh
[CMD] rm -rvf /var/cache/elbe/0193bac3-e521-7746-8603-72f4f87771e5/target/usr/sbin/policy-rc.d
removed '/var/cache/elbe/0193bac3-e521-7746-8603-72f4f87771e5/target/usr/sbin/policy-rc.d'
```

But inside the image it again exists.

```sh
$ cat usr/sbin/policy-rc.d 
#!/bin/sh
exit 101
```

I tried to track down, what causes that, but I'm not sure.

It seems that it is created inside the `_init_` of the `BuildEnv` class (`elbepack/rfs.py`) but unsure why it can "survive" the deletion.

Maybe connected to the following commit (`elbepack/efilesystem.py`):
https://github.com/Linutronix/elbe/commit/459f431604171ce112f99d0195947a021f88b2d4

Is this an expected behavior and if so, is there any way of getting rid of the file via a build definition?

t-8ch, 2024-12-12:
It is not expected and certainly a bug.
Thanks for reporting.

t-8ch, 2024-12-18:
Thanks for the feedback.

This is exactly what `_undo_recursion()` is meant to do.

The diff you posted is exactly the patch 3 of the linked series, no?

martin-29, 2024-12-18:
I did the following:

```xml
<file dst="/usr/sbin/policy-rc.d">
	#!/bin/sh
	exit 0
</file>
```

But it seems that the file is not changed at all.
I created a copy of the file before and after the file command to check if the file command has any effect.

```sh
$ cat usr/sbin/policy-rc.d.orig 
#!/bin/sh
exit 101

$ cat usr/sbin/policy-rc.d.adjusted 
#!/bin/sh
exit 101
```

Does the file overwrite work for you?
Because it may be possible that I made a mistake when applying the patch.
If it worked for you, I would give it tomorrow another try.

t-8ch, 2024-12-19:
So the issue is with `<file>` not actually activating the chroot and therefore the whole excursion mechanism.
It just writes to the target. When using `<command>` it works "correctly".

For the next release I don't want to change it. It's more important to fix the originally leftover file in the target.
Then we can figure out what is the correct behavior for `<file>`.

martin-29, 2024-12-19:
Ok got it. 
Fine for me.

closed 2024-12-19 by t-8ch