Linutronix/elbe #427: Build fails when replacing /etc/hosts by link since 15.0

mrelyb, 2025-01-21:
We need to replace /etc/hosts by a link to a writeable storage, since we keep our rootfs readonly. Up to  v15.0 this was no problem, builds were successfull.
With commit [2984e60](https://github.com/Linutronix/elbe/commit/2984e604c57bfe53ea55b699eead32fe703f7fea) things changed: now we get a build error, when elbe tries to update /etc/hosts in rfs.py at

```
self.rfs.append_file('/etc/hosts',
                             '\n127.0.0.1 localhost'
                             f'\n127.0.1.1 {fqdn} {hostname} elbe-daemon\n')
```

How can I replace /etc/hosts without getting a build error?

log of failing build:

```
[CMD] /usr/sbin/chroot /var/cache/elbe/019488fe-bb3e-726e-aee2-50b2ae52a07d/chroot chpasswd --encrypted
[ERROR]Build failed
Traceback (most recent call last):
  File "/usr/lib/python3/dist-packages/elbepack/asyncworker.py", line 163, in execute
    self.project.build(skip_pkglist=False,
  File "/usr/lib/python3/dist-packages/elbepack/elbeproject.py", line 671, in build
    self.build_cdroms(build_bin, build_sources, cdrom_size, tgt_pkg_lst=tgt_pkgs)
  File "/usr/lib/python3/dist-packages/elbepack/elbeproject.py", line 462, in build_cdroms
    env.seed_etc()
  File "/usr/lib/python3/dist-packages/elbepack/rfs.py", line 333, in seed_etc
    self.rfs.append_file('/etc/hosts',
  File "/usr/lib/python3/dist-packages/elbepack/filesystem.py", line 338, in append_file
    with self.open(path, 'a') as f:
         ^^^^^^^^^^^^^^^^^^^^
  File "/usr/lib/python3/dist-packages/elbepack/filesystem.py", line 85, in open
    return open(self.fname(path), mode, opener=opener)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
FileNotFoundError: [Errno 2] No such file or directory: '/var/cache/elbe/019488fe-bb3e-726e-aee2-50b2ae52a07d/chroot/etc/hosts'
elbe control wait_busy /var/cache/elbe/019488fe-bb3e-726e-aee2-50b2ae52a07d failed!
elbe parselicence failed!
```

t-8ch, 2025-01-23:
@mrelyb There were some bugfixes around this in v15.3. Could you test that?

mrelyb, 2025-01-24:
The above test was acutally with 15.3. Sorry for not mention it...

```
root@e7070abfcbb3:/# elbe --version
elbe v15.3
```

t-8ch, 2025-01-24:
Can you provide a minimal reproducer? XML and command.
Did you bisect the issue to this commit or is it an assumption?

mrelyb, 2025-02-07:
I found the problem: I replaced /etc/hosts by a link during postinst of a package. This led to the above mentioned error.
If I replace /etc/host by a link in finetuning, everything works fine.

Thanks for your patience!

closed 2025-07-31 by t-8ch