Linutronix/elbe #430: Issue with the Built-Using packages from the grub-common package in Ubuntu

Taumille, 2025-02-20:
Hello, I'm trying to build an Ubuntu image with Elbe and I am confronted to a problem regarding the `Built-Using` parameter from Ubuntu archives.
In my specific case, I'm trying to build an Ubuntu Noble image for an x86 platform and so I need the `grub-common` package but in the Packages.xz (link [here](https://archive.ubuntu.com/ubuntu/dists/noble/main/binary-amd64/Packages.xz)) we can see : 
```
[...]
Package: grub-common
Architecture: amd64
Version: 2.12-1ubuntu7
Built-Using: lzo2 (= 2.10-2build3)
[...]
```

As I understand it, the `Built-Using` parameter indicates that grub-common has been built using the lzo2_2.10-2build3 package, downloading the package isn't mandatory to run grub-common on the target however, downloading it is a way to comply with the lzo2 license by providing the source code (cf. https://www.debian.org/doc/debian-policy/ch-relationships.html#additional-source-packages-used-to-build-the-binary-built-using).

This is done in elbe with [thise line](https://github.com/Linutronix/elbe/blob/master/elbepack/aptpkgutils.py#L205)

But this particular case creates an issue as the version 2.10-2build3 of lzo2 isn't available anymore in Ubuntu and has been replace with the version 2.10-2build4. Indeed switching the `include_built_using` variable to False in the code solved the issue.

However I wanted to ask, what is the proper way to solve the issue ? Should I ask the Ubuntu developpers to update the grub-common package to use the latest lzo2 build ? Is there an xml parameter in elbe to switch off the downloading of built-using packages ?

t-8ch, 2025-02-24:
To fulfill source distribution requirements of various Open Source licenses, the package repository should contain all versions of packages which are references by binary packages.
So the Ubuntu repository should contain both versions of the source package. `-build3` and `-build4`, but `-build3` is not available: https://archive.ubuntu.com/ubuntu/dists/noble/main/source/Sources.gz

Taumille, 2025-02-24:
Thank you very much! Indeed, passing `--skip-build-sources` seems to solve the issue. Thank you also for the explanations.

However, I noticed something: the `-build3` version is available in the Ubuntu archives at: https://archive.ubuntu.com/ubuntu/pool/main/l/lzo2/lzo2_2.10-2build3.debian.tar.xz

Is it overgeneralizing to say that all versions referenced in the Package.xz are still available somewhere in the Ubuntu archives? If I understand correctly, it makes sense that they're not always in the `Sources.gz` as we don't want Ubuntu users to download an outdated package.

Would it make sense to search for packages referenced in the `Built-Using` section outside of `Sources.gz` ?

t-8ch, 2025-02-25:
> Is it overgeneralizing to say that all versions referenced in the Package.xz are still available somewhere in the Ubuntu archives?

Everything should be in the archive somewhere. 

> If I understand correctly, it makes sense that they're not always in the Sources.gz as we don't want Ubuntu users to download an outdated package.

Apt would not select the older version by default, so the security argument is weak.
After talking to some experts, it's my opinion that this is a bug in Ubuntu's repository maintenance software. As long as the sources are referenced from binary package currently in the index they should also be available through the same index.
I'm looking for the correct place to report the issue.

> Would it make sense to search for packages referenced in the Built-Using section outside of Sources.gz ?

Only as last resort.

Taumille, 2025-02-25:
> Apt would not select the older version by default, so the security argument is weak.
> After talking to some experts, it's my opinion that this is a bug in Ubuntu's repository maintenance software. As [...]

As the Sources.gz reference `Ubuntu Developers <ubuntu-devel-discuss@lists.ubuntu.com>` as the current maintainer and `Stephen Kitt <skitt@debian.org>` as the original maintainer of the lzo2 package, I would suggest to send them an email to make the `-build3` version available.

t-8ch, 2025-02-25:
> I would suggest to send them an email to make the -build3 version available.

The original maintainer probably has no influence on this. The repository maintenance software should automatically expire and remove old and unreferenced packages. I'm looking for a contact pointer of the repository admins.

t-8ch, 2025-02-25:
Upstream bug: https://bugs.launchpad.net/ubuntu-archive-tools/+bug/2099994
Note: The Debian tooling (dak) handles this correctly.