Linutronix/elbe #431: Issue with root partition

Taumille, 2025-02-25:
Hello,
I just noticed, that since commit [`c5e73aa`](https://github.com/Linutronix/elbe/commit/c5e73aadb421d75809262e96264f63fc974d87e8), the grub installation is broken. 
It seems that grub is trying to mount the loop device used to build the distribution on the target device.

For example here is an extract of the grub boot command generated for the example ` x86_64-pc-hdimg-grub-uefi-bookworm.xml` : 
```
echo ‘Loading Linux 6.1.0-31-amd64 ...'
linux /boot/vmlinuz-6.1.0-31-amd64 root=/dev/loop0p2 ro 
              (Here is the problem ^)
echo ‘Loading initial ramdisk ...'

initrd /boot/initrd.img-6.1.0-31-amd64
```

As this loop device name made sense only for the build machine, the target device can't boot further than the initramfs and gives the error :
```
ALERT ! /dev/loop0p2 does not exist. Dropping to a shell !
```

The issue is due to the script [`/etc/grub.d/10_linux`](https://github.com/olafhering/grub/blob/4dc6166571645780c459dde2cdc1b001a5ec844c/util/grub.d/10_linux.in#L54) called by [`update-grub2`](https://github.com/Linutronix/elbe/blob/master/elbepack/hdimg.py#L175) and in particular to those lines : 
```bash
if ( [ "x${GRUB_DEVICE_UUID}" = "x" ] && [ "x${GRUB_DEVICE_PARTUUID}" = "x" ] ) \
    || ( [ "x${GRUB_DISABLE_LINUX_UUID}" = "xtrue" ] \
	&& [ "x${GRUB_DISABLE_LINUX_PARTUUID}" = "xtrue" ] ) \
    || ( ! test -e "/dev/disk/by-uuid/${GRUB_DEVICE_UUID}" \
	&& ! test -e "/dev/disk/by-partuuid/${GRUB_DEVICE_PARTUUID}" ) \
    || ( test -e "${GRUB_DEVICE}" && uses_abstraction "${GRUB_DEVICE}" lvm ); then
  LINUX_ROOT_DEVICE=${GRUB_DEVICE}
elif [ "x${GRUB_DEVICE_UUID}" = "x" ] \
    || [ "x${GRUB_DISABLE_LINUX_UUID}" = "xtrue" ]; then
  LINUX_ROOT_DEVICE=PARTUUID=${GRUB_DEVICE_PARTUUID}
else
  LINUX_ROOT_DEVICE=UUID=${GRUB_DEVICE_UUID}
fi
```

What I saw is that before the commit, the file `/dev/disk/by-uuid/${GRUB_DEVICE_UUID}` existed and so the if statement returned false and as `GRUB_DEVICE_UUID` was set we were falling in the else statement, which is good because the rootfs partition doesn't change its UUID from the build machine to the target machine.

But now, `update-grub2` can't find  `/dev/disk/by-uuid/${GRUB_DEVICE_UUID}` nor `/dev/disk/by-partuuid/${GRUB_DEVICE_PARTUUID}` so the if statement return true and we're using the partition name as the Linux root device so `/dev/loop0p2` on the build machine and something like `/dev/sda2` on the target machine.

t-8ch, 2025-02-26:
Thanks for the report. I'll try to reproduce it, but I think that we need to wait until udev is done. Not yet sure how to do this in a race-free and correct way.

t-8ch, 2025-02-28:
Should be fixed by https://lists.linutronix.de/pipermail/elbe-devel/2025-February/007655.html
Any feedback is welcome.
It's a bit unfortunate the update-grub2 will silently ignore an explicitly specified `GRUB_DEVICE_UUID` if the device symlink is missing.

closed 2025-03-10 by Taumille

t-8ch, 2025-03-10:
Thanks for the confirmation!