Linutronix/elbe #433: Behavior of .so symlinks in SDK

martin-29, 2025-03-10:
I have a question regarding the ELBE SDK and the behaviour of the symlinks coming from `*-dev` pkgs.
 
Let's say I have in my target pkg-list the package `libgpiod2` and want to have the development files in my SDK.
 
For that I add the `libgpiod-dev` package to the `hostsdk-pkg-list` to have the header files available.
 
Now I would assume that when cross-compiling a project, the shared object can be found by the compiler because of the sysroot option.
 
```sh
echo $CC
aarch64-linux-gnu-gcc --sysroot=/opt/elbe-sdk-aarch64-linux-gnu-test-1.0.0/sysroots/target
```
 
But in `target/` the symlink pointing to the shared object with version info is not there, because this link comes with the `*-dev` package which is only inside `host/`.
 
```sh
# host/
ls -l ./host/usr/lib/x86_64-linux-gnu/libgpiod.so
lrwxrwxrwx 1 root root 17 Jan  5  2023 ./host/usr/lib/x86_64-linux-gnu/libgpiod.so -> libgpiod.so.2.2.2
 
# target/
ls -l ./target/usr/lib/aarch64-linux-gnu/libgpiod*
lrwxrwxrwx 1 root root     20 Jan  5  2023 ./target/usr/lib/aarch64-linux-gnu/libgpiodcxx.so.1 -> libgpiodcxx.so.1.1.2
-rw-r--r-- 1 root root 133056 Jan  5  2023 ./target/usr/lib/aarch64-linux-gnu/libgpiodcxx.so.1.1.2
lrwxrwxrwx 1 root root     17 Jan  5  2023 ./target/usr/lib/aarch64-linux-gnu/libgpiod.so.2 -> libgpiod.so.2.2.2
-rw-r--r-- 1 root root  67512 Jan  5  2023 ./target/usr/lib/aarch64-linux-gnu/libgpiod.so.2.2.2
```
 
I understand that this behaviour comes from the way Debian seperates the packages:
 
*The development package should contain a symlink for the associated shared library without a version number. For example, the libgdbm-dev package should include a symlink from /usr/lib/libgdbm.so to libgdbm.so.3.0.0. This symlink is needed by the linker (ld) when compiling packages, as it will only look for libgdbm.so when compiling dynamically.*
 
*From: https://www.debian.org/doc/debian-policy/ch-sharedlibs.html#development-files*
 
And of course you could resolve that issue quite easily by creating the symlink manually or by fixing the things within CMake etc. 
But would be cool if this simply works out of the box.
 
What do you think about this topic, is this something that could be improved?

t-8ch, 2025-03-10:
If you are building a SDK you should get the -dev packages in the target architecture in the target sysroot.
Otherwise there are also no header files to build against.

t-8ch, 2025-03-10:
> For that I add the libgpiod-dev package to the hostsdk-pkg-list to have the header files available.

Specifically this should not be necessary, but automatic.
Adding it to `hostsdk-pkg-list` is wrong as you want it for the target system.
The `host` sysroot is meant for build tools or compilers.

martin-29, 2025-03-11:
Yes, that with the `hostsdk-pkg-list` makes sense. That was probably only a desperate attempt to solve the issue, after I saw the missing "dependencies".

When you say that we should get the -dev packages in the target sysroot, do you mean that this should work implicitly by having for example `libgpiod2` inside the `pkg-list`?
Because for me that isn't the case.
I would need to add the -dev package explicitly to the `pkg-list`, but I don't want to have the -dev packages inside my final image.

t-8ch, 2025-03-11:
> When you say that we should get the -dev packages in the target sysroot, do you mean that this should work implicitly by having for example libgpiod2 inside the pkg-list?

Correct.

>  Because for me that isn't the case.

Can you provide a reproducer?

martin-29, 2025-03-11:
Ok I checked again the target sysroot and interestingly for most of the packages it seems that the dev packages are installed.

For example, I have `libubootenv-tool` inside my `pkg-list` and here I see the files coming from `libubootenv-dev`.
So here everything looks fine.

```sh
find -name libuboot*
./sysroots/target/usr/lib/aarch64-linux-gnu/libubootenv.so
./sysroots/target/usr/lib/aarch64-linux-gnu/libubootenv.a
./sysroots/target/usr/lib/aarch64-linux-gnu/pkgconfig/libubootenv.pc
./sysroots/target/usr/lib/aarch64-linux-gnu/libubootenv.so.0.3.2
./sysroots/target/usr/lib/aarch64-linux-gnu/libubootenv.so.0
./sysroots/target/usr/include/aarch64-linux-gnu/libuboot.h
```

But at least for `libgpiod2` this isn't the case for me.
Only the files from the pkg `libgpiod2` itself are included.

```sh
find -name libgpiod*
./sysroots/target/usr/lib/aarch64-linux-gnu/libgpiodcxx.so.1.1.2
./sysroots/target/usr/lib/aarch64-linux-gnu/libgpiod.so.2
./sysroots/target/usr/lib/aarch64-linux-gnu/libgpiodcxx.so.1
./sysroots/target/usr/lib/aarch64-linux-gnu/libgpiod.so.2.2.2
```

I had also a short look into the sources and if I understand the process correct ELBE resolves the source package name for each installed binary package and checks if a matching `-dev` package is available.

If that is true, I don't undestand why this doesn't work for `libgpiod`/`libgpiod-dev`.
`libgpiod2` is built from the source `libgpiod`.

We build based on `bookworm` with ELBE 15.5.

t-8ch, 2025-03-11:
> We build based on bookworm with ELBE 15.5.

I tested this with `tests/simple-arm64.xml` which is based on bullseye. That worked fine.
But when switching it to bookworm it is broken for me, too.
Taking a look.

martin-29, 2025-03-11:
I recognized one difference inside the control file between the bullseye and bookworm version.

In bullseye:
```sh
Package: libgpiod2
Source: libgpiod
...
```

In bookworm:
```sh
Package: libgpiod2
Source: libgpiod (1.6.3-1)
...
```

Do you think the version breaks the resolving of the -dev packages?

t-8ch, 2025-03-11:
> Do you think the version breaks the resolving of the -dev packages?

This looks suspicious. Thanks for the pointer.

t-8ch, 2025-03-13:
Could you give the following diff a try? Apply it inside the initvm and then do `systemctl restart python3-elbe-daemon`.

```diff
diff --git a/elbepack/rpcaptcache.py b/elbepack/rpcaptcache.py
index e74d21a7827f..cbff48f41ee9 100644
--- a/elbepack/rpcaptcache.py
+++ b/elbepack/rpcaptcache.py
@@ -126,7 +133,7 @@ class RPCAPTCache(InChRootObject):
                 src_name = pkg.candidate.source_name
                 src_name_lst.append(src_name)
                 version_dict[pkg.name] = pkg.candidate.version
-                version_dict[src_name] = pkg.candidate.version
+                version_dict[src_name] = pkg.candidate.source_version
 
         def mark_install(pkg_lst, suffix):

```

martin-29, 2025-03-15:
Hi, thanks a lot.
That seems to fix the issue.
For `libgpiod2` and other packages I get now automatically the `-dev` package inside the target sysroot.

t-8ch, 2025-03-21:
Patch submitted as https://lists.linutronix.de/pipermail/elbe-devel/2025-March/007686.html

closed 2025-03-31 by t-8ch