Linutronix/elbe #441: v15.6 elbe control set_orig build_dir orig_file fails

erwaelde, 2025-07-29:
Hello,

while working my way to upgrade my build environment from elbe 14.9.3 to 15.6 I possibly found another error. I can build packages of unmodified sources, however, adding a (modified) source file fails like so:

```
$ path/to/.../elbe/v15.6/elbe control set_orig "/var/cache/elbe/01985489-7b91-70e5-a5d9-087635e1eace" "/space/ew/GIT/tse-linux/noissue-prepare-for-trixie/2023/foe-i2c-tools-d12/i2c-tools_4.3.orig.tar.xz"
Server returned error:

Traceback (most recent call last):
  File "/var/cache/elbe/devel/elbepack/daemons/soap/esoap.py", line 67, in call_wrapper
    return super().call_wrapper(ctx)
           ^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/lib/python3/dist-packages/spyne/service.py", line 194, in call_wrapper
    return ctx.function(*args)
           ^^^^^^^^^^^^^^^^^^^
  File "/var/cache/elbe/devel/elbepack/daemons/soap/esoap.py", line 205, in append_upload_orig
    orig_fname = os.path.join(builddir, self.app.pm.get_orig_fname(builddir))
                 ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "<frozen posixpath>", line 90, in join
  File "<frozen genericpath>", line 152, in _check_arg_types
TypeError: join() argument must be str, bytes, or os.PathLike object, not 'NoneType'

```

My python fu is not sufficient for these kind of things. So any help is appreciated.

Cheers,
erwaelde

t-8ch, 2025-07-29:
Thanks for the report. I do have a fix for this specific issue, but I'll need to work on it some more.
Did you also take a look at the all-in-one `elbe initvm submit` based workflow?

erwaelde, 2025-07-29:
Hello, thanks for your reply.

I'm not sure what you mean by `elbe initvm submit` based workflow.

I build .deb packages and images.

Images are using `initvm create` and `initvm ensure` to build an initvm instance or start an existing one. Then `initvm submit` feeds a .xml file (including some archive blob) into that machine and produces the artefacts.

.deb packages are using `initvm create` and `initvm ensure` as well. After that `control list_projects` will check, if the needed pbuilder is already available. If not, `pbuilder create` will create one. `control set_orig` is issued if needed, then `pbuilder build` will do the work.

Both cases are implemented as shell scripts, which will also package and archive artefacts, upload .deb packages to an internal .deb repo (aptly/aptly-api style). This repo is also used as a source repo to the package and image build jobs.

I operate under the assumption that an initvm is a ephemeral thing that must be recreatable at any time, even though I have learned to remove unused or failed projects inside the initvm. I do not actively use the cache facilities inside the initvm. So maybe I'm missing something.

While I'm at it: I have moved pretty much everything, that belongs to the finetuning section into external scripts --- which are then called in finetuning. I prefer a build to fail, if finetuning sees any error. When building for very small systems, I create an image that can run from sdCard. And in a separate step I open/mount said image and copy a hand crafted selection of files over to an empty container, which is converted to an initial ram disk. I find this a lot more manageable than trying to convince elbe to do that for me (I have tried :-) --- that is for your and others information only, there is nothing wrong with `elbe`.

Thanks for your time!
erwaelde

t-8ch, 2025-07-31:
> While I'm at it: I have moved pretty much everything, that belongs to the finetuning section into external scripts --- which are then called in finetuning. I prefer a build to fail, if finetuning sees any error.

The current finetuning should report encountered errors and fail. What is not working for you?

> which is converted to an initial ram disk

elbe can also build cpio images and create images from scratch, without installing a full operating system.
See `examples/x86_64-pc-rescue-busybox-dyn-cpio.xml`. Currently that installs a full glibc, pulling in libgconv, too.
But these can be either finetuned away are avoided by using `busybox-static`. (Sending a patch for that)

(Still working on your packaging workflow.)

t-8ch, 2025-08-01:
> I got tangled up in things, which are not forseen in the xml representation of the finetuning section, paying attention to quoting and < redirection and similar.

Fair enough. You could also use xinclude to let ELBE figure out all of the quoting:

```xml
<ns0:RootFileSystem xmlns:ns0="https://www.linutronix.de/projects/Elbe" ... xmlns:xi="http://www.w3.org/2003/XInclude" >
...
<finetuning>
  <command>
    <xi:include parse="text" href="finetuning.sh" />
  </command>
</finetuning>
...
```

erwaelde, 2025-09-08:
Good hint!
Way down in the trenches the magic variable was unset ... After fixing that my pbuilder .deb-packages builds are all good now.
Thank you for your support.